bbbenji/PMSBLM #2: Helper tool doesn't seem to be working properly?

4nthonylin, 2020-08-13:
I took my bilinear levelling grid and used both the new and old methods and got different results. The old method seems to generate the current values whereas the new one generates invalid ones. 

## bilinear levelling grid
```
      0      1      2      3
 0 +0.074 +0.127 +0.051 -0.010
 1 +0.148 +0.146 +0.125 +0.128
 2 +0.057 +0.131 +0.171 +0.216
 3 -0.044 +0.165 +0.245 +0.329
```

## Old
```

      Raw values:
      -0.19	0.06	0.19
      -0.04	0	0.03
      -0.07	-0.05	-0.15
      
      Degrees:
      137°CCW	43°CW	137°CW
      29°CCW	0	22°CW
      50°CCW	36°CCW	108°CCW

      Fractional Turns:
      2/5CCW	1/10CW	2/5CW
      1/10CCW	0	1/10CW
      1/10CCW	1/10CCW	3/10CCW
```
[image]

## New
```
Raw values:
-0.06 +0.04 +0.10
-0.00 ±0.00 -0.05
-0.11 +1.39 +2.89
Degrees:
40° CCW 28° CW 75° CW
1° CCW 0° 38° CCW
81° CCW 999° CW 2079° CW
Fractional Turns:
1/10 CCW 1/10 CW 1/5 CW 
0 CCW 0 1/10 CCW
1/5 CCW 14/5 CW 29/5 CW
```
[image]

bbbenji, 2020-08-13:
You need to paste the output sans the first line `0 1 2 3`

[image]

I will implement a check for presence of that line and calculate accordingly to avoid this problem in the future.

closed 2020-08-13 by bbbenji

4nthonylin, 2020-08-13:
Ah I see, for some strange reason I did try without the first line and it also gave me results that didn't seem to match with my expectations. Must've copied something extra potentially!

bbbenji, 2020-08-13:
It shouldn't matter now. I pushed some changes to handle such situations 😀

commit 1741b7f7 (2020-08-13): remove `0 1 2 3` if exists. closes #1 closes #2