bbbenji/PMSBLM #3: Improvements

Glademist, 2020-09-03:
Hello, thanks for the great guide. Just finished it. My feedback below that could improve the experience of others.

1 - the 16mm screws are DIN 965 or DIN 7991 and all HEX and CROSS and TORX versions of the screw in size M3x16mm work. I personally used a CROSS DIN 965 M3 x 16 mm without issues. The screw head is flush or slightly below the level of the heatbed. Those can be bought at most hardware stores.

2 - The Hexagonal M3 nut height is really superimportant and should be highlighted that basically ONLY "axial m3 nut" by company Axial works. They build RC cars. There is no DIN number for those. They are custom made for their RC cars. I had to buy them online from an e-shop. The offer of the Axial M3 nut is quite large, most RC eshops sell them.

3 - Probably hardest for me was getting the hose since most of them are 3mm x 7mm. But in the end i found it in a TUNING car e-shop. 

4 - There could be mentioned that at **Installation step 10**, a good guide is to place the spacers beside the silicon hoses and finish screwing just slightly above them. You can avoid a ruler then, the spacers are exactly 6mm (or 6.06 in my case). 

5 - Just before **Leveling Procedure step 5** there should be added "Replace your metalic PEI print sheet". It is not obvious and doesnt work without it. It cost me some time, and introduced uncertainty into the process. There should be mentioned then, that after modifying the screws, the user should replace the PEI sheet and wait for the bed temperature to stabilize on 60. Since if the screwdriving takes long it can drop as low as 56. 

6 - Although i used the 3D printed M3 wrench, i would mention there, that the wrench is slightly bigger and allows a few degrees free movement of the nut inside. So i would either mention to get a metal one, or to first properly "seat" the M3 wrench against the nut to properly brace against the screwdriver movement.

7 - at step **Leveling procedure 7** there should be mentioned, that sometimes in windows, when copying the text output from Pronterface, windows out of some reason dont copy the new numbers with CTRL-C and keep pasting the old ones. So i recommend selecting by mouse and right click COPY. 

Everything else is pretty awesome and i thank you for the guide!

bbbenji, 2020-09-03:
Thanks for the detailed report. I will be sure to mention these things in https://github.com/bbbenji/PMSBLM/blob/master/README.md when I find a moment. Alternatively, a pull request would be great as well.

Glademist, 2020-09-03:
Oh i must learn how to do with GIT. I will try to learn and in case do a
pull request!

bbbenji, 2020-09-04:
The easiest way to start would be to click the edit icon for the specific file you want to edit, in this case https://github.com/bbbenji/PMSBLM/blob/master/README.md

It should take you here: https://github.com/bbbenji/PMSBLM/edit/master/README.md and ask you to fork the repository.

After you save your changes and you feel it's ready to be merged into the original repo, you can click the big green `Create pull request` button on the next page or create it from your fork in the`Pull requests` tab. See here for more: https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request

Of course you can do all of this from the command line, which is a bit more advanced, but definitely worth learning if you are interested. See: https://docs.github.com/en/github/getting-started-with-github/set-up-git

bbbenji, 2020-11-11:
Closed via: https://github.com/bbbenji/PMSBLM/pull/5

closed 2020-11-11 by bbbenji