bbbenji/PMSBLM #4: Why not use the auto-bed-leveling with the minda-probe on the prusa-mini?

ozett, 2020-10-26:
Why not use the auto-bed-leveling with the minda-probe on the prusa-mini?

since the bed-leveling seems a design-bug in the mini (maybe fixed with firmware),
[see this one for the hw-bug](https://forum.prusaprinters.org/forum/assembly-and-first-prints-troubleshooting/brand-new-prusa-mini-z-needs-to-adjust-on-every-new-print/paged/4/), and more on the prusa-forum

why do it all mechanical like your way?

bbbenji, 2020-10-26:
According to Prusa, can't find the source right now, the M/PINDA can compensate for ~.05mm of difference. Some have had luck up to 1.1mm even. But the compensation has its limits. That's where this mod comes into play.

This mod is not required, but it can help in cases when the firmware can't handle irregularity in the print bed. Additionally, I believe it's better to not have to rely on software to fix a hardware problem.

Regarding the "MINDA H/W bug", this mod is not something that will help, and any S/W fix that PRUSA might release, will not be a fix but a workaround. The MINDA is limited in repeatable accuracy due to the missing thermister and this cannot be "fixed" in code without additional hardware. In short, the values returned by the MINDA differ depending on the temperature of the sensor itself. Without knowing the temperature of the sensor, the sensor cannot return an absolute measurement, but a relative one.

ozett, 2020-10-27:
yeah  , thanks.
i had problems with first-layer on my mini and investigated the prusa-forum for this.
there is ongoing discussion about minda...think the hw-mod is a (temporary) solutions.
most people expect the bed-leveling to be correct everytime without manual intervention = out of the box.

thanks for soem way out of  this..

bbbenji, 2020-10-27:
As I mentioned, this is not a fix for the MINDA value discrepancies due to sensor temperature issue. This mod will NOT fix that.

closed 2020-10-27 by bbbenji