bbbenji/PMSBLM #10: Add Klipper SCREWS_TILT_CALCULATE option

weust1979, 2021-12-23:
For those who broke the appendix on their Buddy board and flashed Klipper firmware on it, Octoprint and Pronterface will no longer work.
But, Klipper has SCREWS_TILT_CALCULATE.
It even shows directly how many minutes you must turn the screws (counter) clockwise.

https://www.klipper3d.org/Manual_Level.html#adjusting-bed-leveling-screws-using-the-bed-probe
https://www.klipper3d.org/Config_Reference.html#screws_tilt_adjust
https://www.klipper3d.org/G-Codes.html#bed-screws-tilt-adjust-helper

So having this option added to the documentation would help others too.

The above is thanks to people on the Klipper Discord server for pointing me to the above links.

bbbenji, 2022-02-01:
As I don't run Klipper, I won't be able to document the process. However, feel free to submit a PR with the additions.

bbbenji, 2022-09-28:
The ABL converter has been [rewritten](https://github.com/bbbenji/PMSBLM/commit/1758d81d47be631659f3c0a0564cd0a65707e817) to support Klipper output as well as probably any other printer/firmware here: [bbbenji.github.io/PMSBLM/beta](https://bbbenji.github.io/PMSBLM/beta/)

Note that it is currently in beta, and will eventually replace the current converter here: [bbbenji.github.io/PMSBLM](https://bbbenji.github.io/PMSBLM/)

closed 2022-09-28 by bbbenji