bbbenji/PMSBLM #14: Question

jreed1701, 2022-02-25:
Will the DIN 965 torx screws and nuts  recommended for this procedure also work on an MK3S?

bbbenji, 2022-02-25:
No. However, I believe that you can re-use the original MK3 screws. Unfortunately, that's out of the scope of this repo.

closed 2022-02-25 by bbbenji

jreed1701, 2022-02-25:
Thank you for responding. I had re-used the original MK3 screws doing this procedure on that platform and have noticed they loosen over time. Therefore I was considering torx screws as an alternative.

jreed1701, 2022-02-25:
I failed to say that I am not using nuts on the bottom with the original MK3 screws. The mod works if I make re-leveling the bed a part of maintenance routine. 

I agree. The carriage would be better designed if it had a method to secure the screw.