bbbenji/PMSBLM #16: Right Rear corner to close but bed levling says its Fine

BenHartwich2005, 2022-05-20:
I did the mod and before there were no big troubles but now my right rear corner is waaay to close. But the Probe says its fine but it isn´t 
Any Ideas?)
Things I tried: 
Making the X Axis Straight to the bed 
Flipping the sheet over
Checking for debree
Heres the Rear right Corner:
[image]
Left rear Corner for comparison:
[image]
My Assembly Looks like this:
[image]

BenHartwich2005, 2022-05-20:
Thats what he printer spits out:
 `
0 -0.091 -0.071 -0.026 -0.007
1 -0.089 +0.002 -0.049 -0.013
2 -0.038 +0.022 -0.052 -0.012
3 -0.063 -0.011 -0.005 -0.019
`

BenHartwich2005, 2022-05-20:
[image]

bbbenji, 2022-05-20:
I've seen something similar when the toolhead was depressing the heated bed cable cover. Please check for this and let me know.

bbbenji, 2022-05-20:
By looking closer at the pictures, looks like you are referring to the front right side? Hmm, not sure in this case. Perhaps a faulty SPINDA cable?

BenHartwich2005, 2022-05-20:
I checked it seems to be fine no interruptions

BenHartwich2005, 2022-05-21:
I just tried squaring my Mini and it seens to be better now
https://help.prusa3d.com/article/squaring-your-mini_158518

closed 2022-05-21 by BenHartwich2005