bbbenji/PMSBLM #22: Extend to Prusa MK3S+ ?

rgesswein, 2022-08-21:
Hello!

This is an excellent addition or modification for the printer; I actually implemented following the instructions and results are amazing. Thank you for sharing this with the community.

I'm wondering how hard it can be to extend this project to the Prusa MK3S+ too ?

Thank you!

Rodrigo

bbbenji, 2022-08-21:
I ran across this a year ago. But can't vouch for it: https://www.schweinert.com/silicone-bed-level-mod-prusa-mk3/

bbbenji, 2022-09-28:
The ABL converter has been [rewritten](https://github.com/bbbenji/PMSBLM/commit/1758d81d47be631659f3c0a0564cd0a65707e817) to support MK3 output as well as probably from any other printer/firmware here: https://bbbenji.github.io/PMSBLM/beta/

Note that it is currently in beta, and will eventually replace the current converter here: https://bbbenji.github.io/PMSBLM/

closed 2022-09-28 by bbbenji