bbbenji/PMSBLM #26: Mesh sizes

5tu, 2023-10-12:
The latest MINI alpha firmware uses a 6x6 mesh size instead of 4x4 or 7x7, would it be possible to add this as an option to the 'G29 absolute to relative value converter & plotter' tool? Or even an arbitrary matrix size?

Additionally in 5.1.0-alpha1 `G29 S0` no longer returns the mesh information as per the instructions - `G29 C0 T1` returns the CSV values.

bbbenji, 2023-10-12:
Try this: https://bbbenji.github.io/PMSBLM/beta/ and let me know if it works for you. AFAIR, it should take an arbitrary matrix of values to produce the absolute values. 

Also, could you please post the output of G29 from the latest firmware?

5tu, 2023-10-12:
Thanks, that does appear to correctly plot the values. I assume it shouldn't be inverted though by default?

This is the output when sending `g29`
```
>>> g29
SENDING:G29
Mesh invalidated. Probing mesh.
echo:endstops hit:  Z:1.60
echo:endstops hit:  Z:1.60
echo:endstops hit:  Z:1.60
echo:endstops hit:  Z:1.89
echo:endstops hit:  Z:1.87
echo:endstops hit:  Z:1.87
echo:endstops hit:  Z:2.01
echo:endstops hit:  Z:2.01
echo:endstops hit:  Z:2.01
echo:endstops hit:  Z:2.05
echo:endstops hit:  Z:1.89
echo:endstops hit:  Z:1.88
echo:endstops hit:  Z:1.83
echo:endstops hit:  Z:1.83
echo:endstops hit:  Z:1.83
echo:endstops hit:  Z:1.79
echo:endstops hit:  Z:1.80
echo:endstops hit:  Z:1.80
echo:endstops hit:  Z:1.69
echo:endstops hit:  Z:1.69
echo:endstops hit:  Z:1.69
echo:endstops hit:  Z:1.87
echo:endstops hit:  Z:1.87
echo:endstops hit:  Z:1.87
echo:endstops hit:  Z:1.69
echo:endstops hit:  Z:1.68
echo:endstops hit:  Z:1.69
echo:endstops hit:  Z:1.50
echo:endstops hit:  Z:1.49
echo:endstops hit:  Z:1.49
echo:endstops hit:  Z:1.50
echo:endstops hit:  Z:1.50
echo:endstops hit:  Z:1.50
echo:endstops hit:  Z:0.90
echo:endstops hit:  Z:0.90
echo:endstops hit:  Z:0.90
echo:endstops hit:  Z:1.08
echo:endstops hit:  Z:1.06
echo:endstops hit:  Z:1.45
echo:endstops hit:  Z:1.45
echo:endstops hit:  Z:2.05
echo:endstops hit:  Z:2.06
echo:endstops hit:  Z:2.05
Extrapolating mesh...done
Unified Bed Leveling System v1.01 active
```
This is is when sending `g29 c0 t1`
```
>>> g29 c0 t1 v1
SENDING:G29 C0 T1 V1
Bed Topography Report for CSV:
0.342	0.447	0.892	1.135	1.157	1.207
0.265	0.540	0.815	0.952	0.989	1.025
0.527	0.632	0.737	0.770	0.820	0.870
0.990	0.810	0.630	0.430	0.440	0.450
1.585	0.990	0.395	-0.001	-0.163	-0.324
1.350	1.170	0.160	-0.433	-0.765	-0.755
```
The result is identical when sending `m420 c0 t1` as referenced in the [Marlin documentation](https://marlinfw.org/docs/gcode/M420.html)
```
>>> M420 c0 t1 v1
SENDING:M420 C0 T1 V1
Bed Topography Report for CSV:
0.342	0.447	0.892	1.135	1.157	1.207
0.265	0.540	0.815	0.952	0.989	1.025
0.527	0.632	0.737	0.770	0.820	0.870
0.990	0.810	0.630	0.430	0.440	0.450
1.585	0.990	0.395	-0.001	-0.163	-0.324
1.350	1.170	0.160	-0.433	-0.765	-0.755
Mesh is valid
Storage slot: 0
# of samples: 36
Mean Mesh Height: 0.585729
Standard Deviation: 0.545456
echo:Bed Leveling ON
echo:Fade Height OFF
```
I'm unsure if the mesh should be centred on the mean of the lowest and highest points - which is the `C` flag

bbbenji, 2023-10-12:
@5tu it probably needs to stay inverted. But I'll have to look into this whole issue when o find some spare time. Thanks for letting me know.

biggieticat, 2023-12-13:
@bbbenji @5tu agreed, I think it needs to stay inverted. The old firmware G29 command started the mesh in the right front corner of the bed (when looking at the printer) and worked backwards to get a 4x4 mesh. With the new firmware, G29 makes a 6x6 mesh which starts in the back left corner of the bed. The `G29 C0 `T1`` command gives the necessary table to use in the converter, whereas `G29` alone now will not provide the table

bbbenji, 2024-05-21:
Defaults should be fine. Closing.

closed 2024-05-21 by bbbenji