bbbenji/PMSBLM #32: Add Prusa i3 MK3.5 support please

Investegate, 2025-02-07:
Hey, it would be great if you could a support for the Prusa MK3.5s :)

bbbenji, 2025-02-07:
It should work with Prusa MK3.5s output. Can you post here an example?

Investegate, 2025-02-07:
[image]

However, only the values ​​marked in red should be calculated because that is where the adjustment screws for the heated bed are.

bbbenji, 2025-02-07:
Please paste the output as text. As mentioned, the visualizer should work with what you are showing in your screenshot.

Investegate, 2025-02-07:
`Recv: Bed Topography Report:
Recv: 
Recv:     ( 10,220)                                                                                                                                                      (255,220)
Recv:         0       1       2       3       4       5       6       7       8       9      10      11      12      13      14      15      16      17      18      19      20
Recv: 20 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv: 19 |   .     -0.053    .       .     -0.053    .       .     +0.012    .       .     -0.025    .       .     -0.176    .       .     -0.113    .       .     -0.103    .
Recv:    |
Recv: 18 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv: 17 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv: 16 |   .     +0.017    .       .     +0.040    .       .     +0.042    .       .     +0.017    .       .     -0.003    .       .     -0.045    .       .     -0.068    .
Recv:    |
Recv: 15 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv: 14 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv: 13 |   .     +0.007    .       .     +0.042    .       .     +0.035    .       .     -0.010    .       .     -0.038    .       .     -0.070    .       .     -0.110    .
Recv:    |
Recv: 12 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv: 11 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv: 10 |   .     -0.015    .       .     +0.011    .       .     +0.022    .       .     -0.000    .       .     -0.070    .       .     -0.100    .       .     -0.073    .
Recv:    |
Recv:  9 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv:  8 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv:  7 |   .     +0.010    .       .     +0.020    .       .     +0.030    .       .     -0.005    .       .     -0.025    .       .     -0.063    .       .     -0.135    .
Recv:    |
Recv:  6 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv:  5 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv:  4 |   .     +0.010    .       .     +0.035    .       .     +0.057    .       .     +0.035    .       .     +0.005    .       .     -0.030    .       .     -0.155    .
Recv:    |
Recv:  3 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv:  2 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv:  1 |   .    [-0.015]   .       .     +0.040    .       .     +0.065    .       .     -0.000    .       .     +0.022    .       .     -0.038    .       .     -0.053    .
Recv:    |
Recv:  0 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:         0       1       2       3       4       5       6       7       8       9      10      11      12      13      14      15      16      17      18      19      20
Recv:     ( 10,  -10)                                                                                                                                                      (255,  -10)
Recv: 
Recv: ok
`

Investegate, 2025-02-07:
[7x7UBL-PrusaXL-G29 P1.txt](https://github.com/user-attachments/files/18712073/7x7UBL-PrusaXL-G29.P1.txt)

bbbenji, 2025-02-09:
@Investegate please try the new beta and let me know how it works: https://bbbenji.github.io/PMSBLM/beta/

Just make sure to paste, at MOST, the lines I've highlighted [here](https://github.com/user-attachments/assets/d750be4b-fba1-4441-a340-87e2d677f326)

Attached is an example file: [MAX-7x7UBL-PrusaXL-G29.P1.txt](https://github.com/user-attachments/files/18726508/MAX-7x7UBL-PrusaXL-G29.P1.txt)

Investegate, 2025-02-10:
@bbbenji yep it works, thank you :) 

However, it would be good if only the 3x3 measurement values ​​could be displayed in the stats so that the "Max Difference" value is not influenced by the other 40 measurement points.

bbbenji, 2025-02-10:
@Investegate I won’t implement a feature to display only the 3×3 measurement values because the bed needs to be leveled as one entire plane, and changing the tension on a single screw can impact the bed’s level well beyond that location. Moreover, displaying an accurate visual requires using all the measurement data, not just a limited 3×3 grid, to ensure users see the full picture of how the bed might be warped or tilted across its entire surface.

Investegate, 2025-02-10:
@bbbenji Yes, but that still doesn't make any sense. The heated bed is mounted at 9 points. And at all 9 points the difference should be 0 if possible. The other 40 measured values ​​are useless and only cause confusion. Even with the octoprint plugin "Prusa Leveling Guide" you only get confusing values ​​if you try this in the 7x7 mbl.

bbbenji, 2025-02-10:
The original script was developed for the Prusa Mini, where probe points were never placed directly above the screws. Instead, the approach involved averaging the surrounding probe points and using that data to calculate the required screw rotations. The project has since evolved to accept output from different machines.

As for the MK3.5S, are the probe points located directly above the bed’s mounting points. I suspect not, since that would likely interfere with the S/PINDA readings.

I will need to consider the best way forward, although I’m uncertain if I’ll have time to implement any changes. I think a switch in the UI where the end-user decides, would be best.

Pull requests, however, are always welcome.